RoboticsBrno/ArduinoLearningKitStarter #21: Přidat konektory pro I2C a SPI

yaqwsx, 2018-01-23:
Nikdo si na Arduinu nepamatuje, kde je připojené I2C a kde je SPI. Je to třeba dohledávat a tohle je opruz. Buď chci mít popisky na desce anebo ještě extra pin heady, kde je přímo vyvedené I2C a SPI.

JarekParal, 2018-01-23:
Navrhuji nadefinovat I2C a SPI pro ESP32, tak jak je na klasickém Arduino Uno a přidat popisky na DPS pro Uno pinout.

yaqwsx, 2019-01-02:
> @yaqwsx <https://github.com/yaqwsx>
> Nebylo by lepší dat konektory úplně k sobě, aby se lépe pájeli? Případně
> mezi nimi nechat mezeru jeden pinhead?
> [image: image]
> <https://user-images.githubusercontent.com/6726556/50598334-bb3bd300-0eaa-11e9-8478-ad3b0d527fdf.png>
>
> A proč jsi zvolil pořadí pinů (GND, VCC...)?

Motivací pro mezeru byla snaha o vizuální oddělení. Můžu jí udělat větší.

Pořadí - máme nějaké zažité? Datové piny jsem dal sestupně dle čísel na
Arduinu.

JarekParal, 2019-01-16:
Po rychlém průzkumu konektorů na LCD displejů (většina je má tak jak jsou aktuálně na DPS) a Gyroskopů (praktický všechny to mají trochu jinak) bych konektory ponechal tak jsou a jen bych mezi nimi udělal mezeru o velikosti jednoho pindeadu (lepší pájení a jednotnost).