RoboticsBrno/ArduinoLearningKitStarter #27: Vyřešit připojení pieza na ESP32

yaqwsx, 2018-01-23:
Na ESP32 jsme v aktuální konfiguraci stále uspokojivě nerozpištěli piezo.

CornyjK, 2018-06-18:
Není to dobré řešení, ale fungovalo mi jen s jedním pinem na low a použít funkci ledc (není na AVR). Chtělo by to port knihovny ToneAC.

CornyjK, 2018-06-18:
Navrhuji zavřít, issue přesunut do RoboticsBrno/ArduinoLearningKitStarter-library#11

yaqwsx, 2018-06-18:
ToneAC dělá to, že na půl periody nastaví jednu nožičku na low a druhou na high a pak je prohodí. Díky tomu má piezo dvakrát tak velký rozkmit a je hlasitější. 

To samé by mělo být možné udělat na ESPčku - nevím, jestli ledc umí invertovat pin nebo ne. Pokud ano, tak stačí nastavit oběma nožičkám PWM na 50 % a jeden pin invertovat. Pokud ne, tak by to mohly umět timery pro řízení motorů.