RoboticsBrno/ArduinoLearningKitStarter #31: Aktualizovat obrázky v readme.md

CornyjK, 2018-06-15:
Současné obrázky nejdou s dobou...

CornyjK, 2018-06-15:
Zároveň vytvořit obrázky i pro [wiki](https://github.com/RoboticsBrno/ArduinoLearningKitStarter/wiki/Nastavení-desky-ALKS).

JarekParal, 2018-06-18:
Na wiki můžeš použít tyhle obrázky/fotky: http://robotikabrno.cz/robotika-brno/navody/arduino/alks-tutorial-osazovani

Částečně by šli použít asi i na README.md.

Nové fotky nafotíme snad během tábora.

CornyjK, 2018-06-18:
Asi do wiki hodím i návod osazování.

JarekParal, 2018-06-18:
Souhlas. Přemýšlel jsem ještě, jestli celý návod nepřesunout na GitHub? @CornyjK

CornyjK, 2018-06-18:
Návod osazování?

CornyjK, 2018-06-18:
@JarekParal 
Je osazovací návod ve správném znění (nejsou v něm chyby nebo změny)?

yaqwsx, 2018-06-18:
Nechceš vyrobit i obrázek s popisky pomocí https://github.com/yaqwsx/PcbDraw ?

JarekParal, 2019-06-03:
> Nechceš vyrobit i obrázek s popisky pomocí https://github.com/yaqwsx/PcbDraw ?

Ping @CornyjK ?

CornyjK, 2019-06-03:
> 
> 
> > Nechceš vyrobit i obrázek s popisky pomocí https://github.com/yaqwsx/PcbDraw ?
> 
> Ping @CornyjK ?

Teď na to zrovna nemám čas.