RoboticsBrno/ArduinoLearningKitStarter #32: Vyřešit otevírání

CornyjK, 2018-06-18:
Při otevření ukáže:

```
The following libraries were not found:

robotarna_arduino-rescue

\home\xmrazek7\kicad\lib\kicad-library\library\Zilog

\home\xmrazek7\kicad\lib\kicad-library\library\xilinx

\home\xmrazek7\kicad\lib\kicad-library\library\Xicor

\home\xmrazek7\kicad\lib\kicad-library\library\Worldsemi

\home\xmrazek7\kicad\lib\kicad-library\library\video

\home\xmrazek7\kicad\lib\kicad-library\library\valves

\home\xmrazek7\kicad\lib\kicad-library\library\ttl_ieee

\home\xmrazek7\kicad\lib\kicad-library\library\transistors

\home\xmrazek7\kicad\lib\kicad-library\library\transf

\home\xmrazek7\kicad\lib\kicad-library\library\texas

\home\xmrazek7\kicad\lib\kicad-library\library\switches

\home\xmrazek7\kicad\lib\kicad-library\library\supertex

\home\xmrazek7\kicad\lib\kicad-library\library\stm32

\home\xmrazek7\kicad\lib\kicad-library\library\stm8

\home\xmrazek7\kicad\lib\kicad-library\library\siliconi

\home\xmrazek7\kicad\lib\kicad-library\library\silabs

\home\xmrazek7\kicad\lib\kicad-library\library\sensors

\home\xmrazek7\kicad\lib\kicad-library\library\rfcom

\home\xmrazek7\kicad\lib\kicad-library\library\relays

\home\xmrazek7\kicad\lib\kicad-library\library\regul

\home\xmrazek7\kicad\lib\kicad-library\library\references

\home\xmrazek7\kicad\lib\kicad-library\library\pspice

\home\xmrazek7\kicad\lib\kicad-library\library\Power_Management

\home\xmrazek7\kicad\lib\kicad-library\library\powerint

\home\xmrazek7\kicad\lib\kicad-library\library\power

\home\xmrazek7\kicad\lib\kicad-library\library\philips

\home\xmrazek7\kicad\lib\kicad-library\library\Oscillators

\home\xmrazek7\kicad\lib\kicad-library\library\opto

\home\xmrazek7\kicad\lib\kicad-library\library\onsemi

\home\xmrazek7\kicad\lib\kicad-library\library\nxp_armmcu

\home\xmrazek7\kicad\lib\kicad-library\library\nordicsemi

\home\xmrazek7\kicad\lib\kicad-library\library\msp430

\home\xmrazek7\kicad\lib\kicad-library\library\motorola

\home\xmrazek7\kicad\lib\kicad-library\library\motor_drivers

\home\xmrazek7\kicad\lib\kicad-library\library\microchip_pic32mcu

\home\xmrazek7\kicad\lib\kicad-library\library\microchip_pic18mcu

\home\xmrazek7\kicad\lib\kicad-library\library\microchip_pic16mcu

\home\xmrazek7\kicad\lib\kicad-library\library\microchip_pic12mcu

\home\xmrazek7\kicad\lib\kicad-library\library\microchip_pic10mcu

\home\xmrazek7\kicad\lib\kicad-library\library\microchip_dspic33dsc

\home\xmrazek7\kicad\lib\kicad-library\library\microchip

\home\xmrazek7\kicad\lib\kicad-library\library\microcontrollers

\home\xmrazek7\kicad\lib\kicad-library\library\memory

\home\xmrazek7\kicad\lib\kicad-library\library\maxim

\home\xmrazek7\kicad\lib\kicad-library\library\logo

\home\xmrazek7\kicad\lib\kicad-library\library\linear

\home\xmrazek7\kicad\lib\kicad-library\library\Lattice

\home\xmrazek7\kicad\lib\kicad-library\library\ir

\home\xmrazek7\kicad\lib\kicad-library\library\interface

\home\xmrazek7\kicad\lib\kicad-library\library\intel

\home\xmrazek7\kicad\lib\kicad-library\library\hc11

\home\xmrazek7\kicad\lib\kicad-library\library\graphic

\home\xmrazek7\kicad\lib\kicad-library\library\gennum

\home\xmrazek7\kicad\lib\kicad-library\library\ftdi

\home\xmrazek7\kicad\lib\kicad-library\library\ESD_Protection

\home\xmrazek7\kicad\lib\kicad-library\library\elec-unifil

\home\xmrazek7\kicad\lib\kicad-library\library\dsp

\home\xmrazek7\kicad\lib\kicad-library\library\display

\home\xmrazek7\kicad\lib\kicad-library\library\diode

\home\xmrazek7\kicad\lib\kicad-library\library\digital-audio

\home\xmrazek7\kicad\lib\kicad-library\library\device

\home\xmrazek7\kicad\lib\kicad-library\library\dc-dc

\home\xmrazek7\kicad\lib\kicad-library\library\cypress

\home\xmrazek7\kicad\lib\kicad-library\library\contrib

\home\xmrazek7\kicad\lib\kicad-library\library\conn

\home\xmrazek7\kicad\lib\kicad-library\library\cmos_ieee

\home\xmrazek7\kicad\lib\kicad-library\library\cmos4000

\home\xmrazek7\kicad\lib\kicad-library\library\brooktre

\home\xmrazek7\kicad\lib\kicad-library\library\bbd

\home\xmrazek7\kicad\lib\kicad-library\library\battery_management

\home\xmrazek7\kicad\lib\kicad-library\library\audio

\home\xmrazek7\kicad\lib\kicad-library\library\atmel

\home\xmrazek7\kicad\lib\kicad-library\library\analog_switches

\home\xmrazek7\kicad\lib\kicad-library\library\analog_devices

\home\xmrazek7\kicad\lib\kicad-library\library\Altera

\home\xmrazek7\kicad\lib\kicad-library\library\adc-dac

\home\xmrazek7\kicad\lib\kicad-library\library\actel

\home\xmrazek7\kicad\lib\kicad-library\library\ac-dc

\home\xmrazek7\kicad\lib\kicad-library\library\74xx

\home\xmrazek7\kicad\lib\kicad-library\library\74xgxx

\home\xmrazek7\kicad\lib\downloaded\shield_arduino

C:\Program Files\KiCad\share\kicad\library\graphic

C:\Program Files\KiCad\share\kicad\library\logo

\home\xmrazek7\kicad\yaqwsx_KiCAD_lib\symbols\yaqwsx
```

JarekParal, 2018-10-20:
@yaqwsx Lze v Kicad dělat nějak jinak reference na knihovny než do nějakých uživatelských složek? Jde přikládat knihovny přímo k projektu a referencovat je relativně vůči projektu?

closed 2018-10-30 by CornyjK

JarekParal, 2018-10-30:
Jak jsi to vyřešil @CornyjK?

CornyjK, 2018-10-30:
@JarekParal v `CornyjK-WIP`.