RoboticsBrno/ArduinoLearningKitStarter #38: Arduino library manager

TomasRoj, 2019-04-15:
Čau,

Narazil jsem tady na vás repo a líbí se mi ten nápad! Chci se zeptat jestli je knihovna dostupná na arduino library manager. Pokud ne můžu obstarat :wink:

TomasRoj, 2019-04-15:
Případně ještě přidat nějaké příklady

JarekParal, 2019-04-15:
Ahoj,

pokud myslíš [ArduinoLearningKitStarter-library](https://github.com/RoboticsBrno/ArduinoLearningKitStarter-library), tedy knihovnu k výukové desce ALKS, tak ta je v Arduino Library Manageru dostupná: 
[image]

Pokud budeš mít ale chuť, tak určitě uvítáme PR do [README.md](https://github.com/RoboticsBrno/ArduinoLearningKitStarter-library/blob/master/README.md)  s touto informací.

Co se týče příkladů, tak máme 3 základní [v repozitáři](https://github.com/RoboticsBrno/ArduinoLearningKitStarter-library/tree/master/examples) a taktéž uvítáme PR s dalšími :-).

bsaid, 2019-04-15:
Ahoj, za vytvoření příkladů budeme jenom rádi.

TomasRoj, 2019-04-15:
Ok, teď jdu na to. Jsem rád pokud mohu pomoci arduino komunitě 🔥👌

TomasRoj, 2019-04-15:
jo a ještě. Na jekém pinu na shieldu je bzučák?

JarekParal, 2019-04-15:
https://github.com/RoboticsBrno/ArduinoLearningKitStarter-library/blob/9704050753575767df3dff789c11dc6c3e122450/README.md#piezo

closed 2019-04-15 by TomasRoj